attify/firmware-analysis-toolkit #78: Can't install FAT

asabhish, 2022-10-12:
When I run the setup.sh script, the binwalk installation encounters an error "**fatal: Need to specify how to reconcile divergent branches**". I have installed binwalk separately via the apt repository and it works correctly. This looks like some git pull issue but I am not totally sure. How to resolve this? Attaching the screenshot of the tail-end of the installation log of FAT below:

[image]

asabhish, 2022-10-12:
I could install FAT by commenting out the setup script portion which was installing binwalk tool. And I setup binwalk separately via apt repository. After setting the configuring firmadyne as per the instructions in FAT installation, now I am getting an error while creating the db in PostgreSQL. When I type in the command "**sudo -u postgres createdb -O firmadyne firmware**", the error I get is "**could not change directory to <my firmadyne directory path here>: Permission denied**" ! Any input here as to what is the solution here ? Thanks

extremecoders-re, 2022-10-13:
The installation issue should be fixed with the latest commit.

commit f654856b (2022-10-13): Apply sasquatch patch manually, closes #78
closed 2022-10-13 by extremecoders-re