attify/firmware-analysis-toolkit #79: no route to host

azraelxuemo, 2022-11-09:
I use this firmware
ftp://ftp2.dlink.com/PRODUCTS/DIR-300/REVA/DIR-300_REVA_FIRMWARE_v1.06B05_WW.zip
[image]
[image]
[image]
How to fix it

extremecoders-re, 2022-11-09:
This happens due the bridge not being properly setup within the emulated firmwate.

Run this commands within firmadyne console and it should be reachable from host.

```
brctl show
brctl delif br0 eth2.0
ifconfig eth0 up
brcrtl addif br0 eth0
```

That is you need to delete whatever interface is currently attached on `br0` and then add the `eth0` interface.

azraelxuemo, 2022-11-10:
Thank you very much for your help. It's really solved. Thank you for your kind answer. I hope this issue can help other people with the same confusion

closed 2022-11-10 by azraelxuemo