attify/firmware-analysis-toolkit #81: setup.sh: Consider not patching firmadyne to change interpreter which is standard already

ddorando, 2022-12-21:
https://github.com/attify/firmware-analysis-toolkit/blob/master/setup.sh#L36

Since python2 is EOL anyways (https://www.python.org/doc/sunset-python-2/), it may be worth considering removing this patch to sorta clean-up the firmadyne installation of non-upstream changes.

extremecoders-re, 2022-12-21:
The patch is in place to support Ubuntu 18.04 which defaults to Python2 for the `python` command. 18.04 goes out of support on April and we should be good to remove the patch then. Besides explicitly referring to Python 3 doesn't harm IMO :)

ddorando, 2022-12-21:
Yeah, this is either removing steps which technically aren't needed opposed to

ddorando, 2022-12-21:
Yeah.
This is either about removing a patch which isn't really needed after Ubuntu 18.04 goes EOL (and thus making it kinda redundant to leave it in), or keeping the explicitness that already is there.
In case there isn't any OS that FAT should support supporting python 2 I'd personally suggest to remove it. Just to save a few lines of code and get that tiny bit closer to plain upstream firmadyne.

I know this isn't doing much of anything at this point right now, though IMHO we should keep as close to plain firmadyne as possible to avoid any possible confusion.

adi0x90, 2026-09-25:
I agree with the reasoning here: keeping fewer local patches would make the Firmadyne setup easier to follow.

This particular patch is gone in FAT 2. The new installer doesn't clone Firmadyne or rewrite its interpreter lines. FAT's own Python 3 requirements are in the [installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md).

Closing this since the setup step you wanted to remove is no longer there.

closed 2026-09-25 by adi0x90