attify/firmware-analysis-toolkit #82: fat.py: 'warning: fgrep is obsolescent; using grep -F'

ddorando, 2022-12-21:
When trying to analyze a firmware file, I receive an 'Error: Invalid architecture!' pexpect error.
This is a direct result of my fgrep "binary", which FAT uses, being just a wrapper for 'grep -F' which is additionally outputs said warning first.
The warning is being picked up by FAT as 'architecture'.

```
[+] Firmware: shark-upgrade.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: warning: fgrep is obsolescent; using grep -F
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "/home/rmira/Desktop/firmware-analysis-toolkit/./fat.py", line 172, in <module>
    main()
  File "/home/rmira/Desktop/firmware-analysis-toolkit/./fat.py", line 167, in main
    infer_network(arch, image_id, qemu_dir)
  File "/home/rmira/Desktop/firmware-analysis-toolkit/./fat.py", line 113, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3.10/site-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3.10/site-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3.10/site-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f0ce2fa2dd0>
command: /home/rmira/Desktop/firmadyne/scripts/inferNetwork.sh
args: ['/home/rmira/Desktop/firmadyne/scripts/inferNetwork.sh', '1', 'warning: fgrep is obsolescent; using grep -F']
buffer (last 100 chars): b''
before (last 100 chars): b'Error: Invalid architecture!\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 16861
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'
```

ddorando, 2022-12-21:
[Filed a PR upstream](https://github.com/firmadyne/firmadyne/pull/205)

extremecoders-re, 2022-12-21:
Thanks!
This will be automatically pulled in by fat on new installs.

closed 2022-12-21 by extremecoders-re