attify/firmware-analysis-toolkit #83: Given No IP

yhsnihao, 2022-12-29:
Hey there,
I used Attify OS v3.0，and the firmwork is download from https://www.zhiwanyuzhou.com/download/Firmware/Router/Xiaomi/%E5%B0%8F%E7%B1%B3%E8%B7%AF%E7%94%B1%E5%99%A84A%E5%8D%83%E5%85%86%E7%89%88%EF%BC%88R4A%EF%BC%89/
when I run it like following:
`iot@attifyos ~/t/firmware-analysis-toolkit> 
./fat.py ~/Desktop/Route/miwifi_r4a_firmware_72d65_2.28.62.bin 

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: miwifi_r4a_firmware_72d65_2.28.62.bin
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu`

There's no Network interfaces given.
I had referenced #https://github.com/attify/firmware-analysis-toolkit/issues/46，and increased the timeout value to 180 in scripts/inferNetwork.sh
It doesn't work!
I check the file: firmadyne/scratch/2/qemu.initial.serial.log

jheysel-r7, 2022-12-30:
I'm facing the same issue trying to run the following firmware:  [VMG1312-T20B](https://www.zyxel.com/global/en/support/download?model=VMG1312-T20B:)
```
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] All set! Press ENTER to run the firmware...
```
I also tried increasing the timeout in `inferNetwork.sh` which didn't work for me either.

extremecoders-re, 2023-01-04:
@yhsnihao Just tested on a Ubuntu 20.04 VM. It's working fine and also accessible from the browser.

[image]

[image]
Tunneled with localhost.run.

Contents of run.sh

```sh
#!/bin/bash

set -u

ARCHEND=mipsel
IID=2

if [ -e ./firmadyne.config ]; then
    source ./firmadyne.config
elif [ -e ../firmadyne.config ]; then
    source ../firmadyne.config
elif [ -e ../../firmadyne.config ]; then
    source ../../firmadyne.config
else
    echo "Error: Could not find 'firmadyne.config'!"
    exit 1
fi

IMAGE=`get_fs ${IID}`
KERNEL=`get_kernel ${ARCHEND}`
QEMU=`get_qemu ${ARCHEND}`
QEMU_MACHINE=`get_qemu_machine ${ARCHEND}`
QEMU_ROOTFS=`get_qemu_disk ${ARCHEND}`
WORK_DIR=`get_scratch ${IID}`

TAPDEV_0=tap${IID}_0
HOSTNETDEV_0=${TAPDEV_0}
echo "Creating TAP device ${TAPDEV_0}..."
sudo tunctl -t ${TAPDEV_0} -u ${USER}

echo "Bringing up TAP device..."
sudo ip link set ${HOSTNETDEV_0} up
sudo ip addr add 192.168.31.2/24 dev ${HOSTNETDEV_0}

echo "Adding route to 192.168.31.1..."
sudo ip route add 192.168.31.1 via 192.168.31.1 dev ${HOSTNETDEV_0}

function cleanup {
    pkill -P $$
    
echo "Deleting route..."
sudo ip route flush dev ${HOSTNETDEV_0}

echo "Bringing down TAP device..."
sudo ip link set ${TAPDEV_0} down

echo "Deleting TAP device ${TAPDEV_0}..."
sudo tunctl -d ${TAPDEV_0}

}

trap cleanup EXIT

echo "Starting firmware emulation... use Ctrl-a + x to exit"
sleep 1s

 ${QEMU} -m 256 -M ${QEMU_MACHINE} -kernel ${KERNEL} \
    -drive if=ide,format=raw,file=${IMAGE} -append "root=${QEMU_ROOTFS} console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0" \
    -nographic \
    -netdev tap,id=nettap0,ifname=${TAPDEV_0},script=no -device e1000,netdev=nettap0 -netdev socket,id=net1,listen=:2001 -device e1000,netdev=net1 -netdev socket,id=net2,listen=:2002 -device e1000,netdev=net2 -netdev socket,id=net3,listen=:2003 -device e1000,netdev=net3 | tee ${WORK_DIR}/qemu.final.serial.log
```

yhsnihao, 2023-01-09:
@extremecoders-re Thank you.
I build a clean Ubuntu 22.04 VM，It worked!
[image]

But there's something wrong on Ubuntu 20.04.
[image]
I think I may need more research.

closed 2023-01-12 by yhsnihao