attify/firmware-analysis-toolkit #84: The address is always at 0.0.0.0

Abib-web, 2023-01-03:
Hello,
 I follow your online course, and each time I run the fat.py script on the dlink firmware, I always have the address 0.0.0.0 instead of 192.168.0.1 how to solve this problem

extremecoders-re, 2023-01-04:
Please share a screenshot. Also try in a clean Ubuntu VM.

Abib-web, 2023-01-04:
thank you for your response, but i have change my version of attifyos so i have anoters issues,

this message is normal ? but i cant run web page

```
firmadyne git:(master) ✗ ./fat.py 

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : Dlink_firmware.bin
Enter the brand of the firmware : Dlink
Dlink_firmware.bin
Now going to extract the firmware. Hold on..
/home/oit/tools/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "Dlink_firmware.bin" images 
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Tar2DB
Creating Image
Executing command 

sudo /home/oit/tools/firmadyne/scripts/makeImage.sh 1
Password for user firmadyne: 
Device contains neither a valid DOS partition table, nor Sun, SGI or OSF disklabel
Building a new DOS disklabel with disk identifier 0xfa790078.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
Building a new DOS disklabel with disk identifier 0xa7791a0c.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.9 (4-Feb-2014)
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 3318
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: [('br0', '192.168.0.1')]
Done!

Running the firmware finally : 
WARNING:  Could not open /proc/net/vlan/config.  Maybe you need to load the 8021q module, or maybe you are not using PROCFS??
```

Abib-web, 2023-01-04:
If i use if I enter the enter key

```
Running the firmware finally : 
WARNING:  Could not open /proc/net/vlan/config.  Maybe you need to load the 8021q module, or maybe you are not using PROCFS??

Creating TAP device tap1...
Set 'tap1' persistent and owned by uid 0
Initializing VLAN...
Added VLAN with VID == 0 to IF -:tap1:-
Bringing up TAP device...
Adding route to 192.168.0.1...
Starting emulation of firmware... Done!
The emulated firmware may not be accessible while booting.
Press any key to destroy the network and shutdown emulation.
Deleting route...
Bringing down TAP device...
Removing VLAN...
Removed VLAN -:tap1.0:-
Deleting TAP device tap1... Set 'tap1' nonpersistent
Done!

```

extremecoders-re, 2023-01-04:
Looks like you are using an old version (0.1) of firmware-analysis-toolkit. The interface of the current version (0.3) is different.
Please try with the latest version of FAT on a Ubuntu 20.04 VM.

Abib-web, 2023-01-06:
still the same problem, the address is still 0x000000 and the output on the terminal is the same.

The ping does not work on 192.168.0.1 but it works on 192.168.0.2 however neither of these two addresses is accessible on the browser

[image]

adi0x90, 2026-09-25:
The missing `/proc/net/vlan/config` entry is inside the guest. I'd look at its networking setup as well as the host connection.

FAT 2's native `qemu-direct` system mode uses port forwarding without the old TAP-and-route setup. It still needs guest networking that works with the firmware's VLAN and interface requirements.

Please share the D-Link model/version, backend, serial log, and mapped endpoint from a v2 run. I need to see whether the guest brings up its network and starts the web service before calling this resolved.

adi0x90, 2026-09-26:
The local `Dlink_firmware.bin` sample now extracts successfully after [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108). All 990 regular files and symlink targets matched independent reference extraction. These extraction checks used case-sensitive APFS.

The tested image has SHA-256 `632f8bdd3fc0e8415d490f7581ec13f16aa293ae412e793648fd0d3351b70fe0`. Please compare that with your file so I can confirm whether the result applies to your image.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

I'm keeping this open for the network/browser-access problem. If it remains after a fresh extraction and emulation attempt, please include `fat --version` output, backend, and serial log. The current result verifies extraction; guest networking still needs a runtime check.