attify/firmware-analysis-toolkit #87: fat.py and permission error

torabi12, 2023-02-07:
Could you please help someone about this issue?
[image]

The same command used to work on my side but now it does not start.
I use Ubuntu-Mate 22.04.1 with latest packages.

extremecoders-re, 2023-02-07:
Please run `./clean.py` first and delete the contents of the `images` directory if it isn't empty.
Then you can try running fat.py again.

torabi12, 2023-02-07:
I run ./reset.py and I deleted the contents of the images folder but I got the same:
[image]

I have no idea what has changed and caused this error. :(

torabi12, 2023-02-07:
I tried again with sudo and not it is OK:
[image]

and I was able to log into the web page of the firmware.

Thank you very much for your support!

extremecoders-re, 2023-02-07:
Great to know that its working.

However fat.py shouldn't need sudo to run, it means the images directory is somehow owned by root.

closed 2023-03-20 by extremecoders-re