attify/firmware-analysis-toolkit #88: Error while extracting RAX30 firmware

y198nt, 2023-03-15:
can anyone help me with this, this error occurs when I try to using fat with RAX30 router firmware every version
[+] Firmware: RAX30-V1.0.9.92_1.img
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/home/browser/working_station/iot/firmware-analysis-toolkit/./fat.py", line 173, in <module>
    main()
  File "/home/browser/working_station/iot/firmware-analysis-toolkit/./fat.py", line 161, in main
    image_id = run_extractor(args.firm_path)
  File "/home/browser/working_station/iot/firmware-analysis-toolkit/./fat.py", line 55, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fae3cf953f0>
command: /usr/bin/sudo
args: ['/usr/bin/sudo', '--', '/home/browser/working_station/iot/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py', '-np', '-nk', '../CVE-2023-24749/RAX30-V1.0.9.92_1.img', '/home/browser/working_station/iot/firmware-analysis-toolkit/firmadyne/images']
buffer (last 100 chars): b''
before (last 100 chars): b'24749/RAX30-V1.0.9.92_1.img\r\n>> MD5: 49241d7f2502d60e14e8f6005116fe3d\r\n>> Skipping: image/g3fax...\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 4803
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Tag: '

extremecoders-re, 2023-03-16:
This is likely due to a bug in extractor.py in firmadyne.

As a workaround, you can first manually extract the img using binwalk and then run fat.py on the extracted squashfs.

```
$ binwalk -e ./RAX30-V1.0.9.92_1.img

$ ./fat.py _RAX30-V1.0.9.92_1.img.extracted/777EAC.squashfs
```

y198nt, 2023-03-17:
yeah it was working but I can find any Network interfaces
                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: 777EAC.squashfs
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu

y198nt, 2023-03-17:
this is qemu.initial.serial.log
[    0.453858] [<c018e768>] (kset_register) from [<c0208e74>] (__class_register+0xa8/0x198)
[    0.454376] [<c0208e74>] (__class_register) from [<c0208fa4>] (__class_create+0x40/0x70)
[    0.454920] [<c0208fa4>] (__class_create) from [<c01add28>] (register_devfs_stubs+0x314/0xbb4)
[    0.456380] [<c01add28>] (register_devfs_stubs) from [<c05d7a50>] (init_module+0x28/0xa4)
[    0.458865] [<c05d7a50>] (init_module) from [<c00095cc>] (do_one_initcall+0x104/0x1b4)
[    0.459881] [<c00095cc>] (do_one_initcall) from [<c05c5d08>] (kernel_init_freeable+0xf0/0x1b0)
[    0.461334] [<c05c5d08>] (kernel_init_freeable) from [<c040f04c>] (kernel_init+0x8/0xe4)
[    0.462386] [<c040f04c>] (kernel_init) from [<c0016ce8>] (ret_from_fork+0x14/0x2c)
[    0.463070] ---[ end trace 90832bdce137094d ]---
[    0.463473] ------------[ cut here ]------------
[    0.464013] WARNING: CPU: 0 PID: 1 at /mnt/data/sources/linux/lib/kobject.c:240 kobject_add_internal+0x240/0x2ac()
[    0.464632] kobject_add_internal failed for gpio with -EEXIST, don't try to register things with the same name in the same directory.
[    0.465353] Modules linked in:
[    0.465647] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.1.17+ #10
[    0.466113] Hardware name: Generic DT based system
[    0.466445] [<c001c8dc>] (unwind_backtrace) from [<c0019c70>] (show_stack+0x10/0x14)
[    0.466970] [<c0019c70>] (show_stack) from [<c00248d4>] (warn_slowpath_common+0x80/0xa8)
[    0.467836] [<c00248d4>] (warn_slowpath_common) from [<c0024928>] (warn_slowpath_fmt+0x2c/0x3c)
[    0.468421] [<c0024928>] (warn_slowpath_fmt) from [<c018e5f4>] (kobject_add_internal+0x240/0x2ac)
[    0.468975] [<c018e5f4>] (kobject_add_internal) from [<c018e768>] (kset_register+0x1c/0x44)
[    0.469547] [<c018e768>] (kset_register) from [<c0208e74>] (__class_register+0xa8/0x198)
[    0.470058] [<c0208e74>] (__class_register) from [<c0208fa4>] (__class_create+0x40/0x70)
[    0.471252] [<c0208fa4>] (__class_create) from [<c01add28>] (register_devfs_stubs+0x314/0xbb4)
[    0.472025] [<c01add28>] (register_devfs_stubs) from [<c05d7a50>] (init_module+0x28/0xa4)
[    0.472594] [<c05d7a50>] (init_module) from [<c00095cc>] (do_one_initcall+0x104/0x1b4)
[    0.473103] [<c00095cc>] (do_one_initcall) from [<c05c5d08>] (kernel_init_freeable+0xf0/0x1b0)
[    0.476476] [<c05c5d08>] (kernel_init_freeable) from [<c040f04c>] (kernel_init+0x8/0xe4)
[    0.477092] [<c040f04c>] (kernel_init) from [<c0016ce8>] (ret_from_fork+0x14/0x2c)
[    0.477606] ---[ end trace 90832bdce137094e ]---
[    0.478024] firmadyne: Cannot create device class: gpio!
[    0.480595] firmadyne: Cannot register character device: watchdog, 0xa, 0x82!
[    0.481968] firmadyne: Cannot register character device: wdt, 0xfd, 0x0!
[    0.586797] PCI host bridge /pcie@10000000 ranges:
[    0.587730]    IO 0x3eff0000..0x3effffff -> 0x00000000
[    0.588289]   MEM 0x10000000..0x3efeffff -> 0x10000000
[    0.588648]   MEM 0x8000000000..0xffffffffff -> 0x8000000000
[    0.589709] pci-host-generic 4010000000.pcie: resource collision: [mem 0x00000000-0xffffffff] conflicts with /pl011@9000000 [mem 0x09000000-0x09000fff]
[    0.593651] pci-host-generic: probe of 4010000000.pcie failed with error -16
[    0.614916] Non-volatile memory driver v1.3
[    0.655157] brd: module loaded
[    0.677985] loop: module loaded
[    0.699367]  vda: vda1
[    0.716753] 0.flash: Found 2 x16 devices at 0x0 in 32-bit bank. Manufacturer ID 0x000000 Chip ID 0x000000
[    0.717793] Intel/Sharp Extended Query Table at 0x0031
[    0.718987] Using buffer write method
[    0.719896] erase region 0: offset=0x0,size=0x40000,blocks=256
[    0.729623] 0.flash: Found 2 x16 devices at 0x0 in 32-bit bank. Manufacturer ID 0x000000 Chip ID 0x000000
[    0.731213] Intel/Sharp Extended Query Table at 0x0031
[    0.732563] Using buffer write method
[    0.732951] erase region 0: offset=0x0,size=0x40000,blocks=256
[    0.733383] Concatenating MTD devices:
[    0.733650] (0): "0.flash"
[    0.733840] (1): "0.flash"
[    0.734017] into device "0.flash"
[    0.767327] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.768834] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.769844] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.771947] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.773137] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.776454] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.777647] nand: device found, Manufacturer ID: 0x98, Chip ID: 0x39
[    0.778132] nand: Toshiba NAND 128MiB 1,8V 8-bit
[    0.778776] nand: 128 MiB, SLC, erase size: 16 KiB, page size: 512, OOB size: 16
[    0.781507] flash size: 128 MiB
[    0.782100] page size: 512 bytes
[    0.782592] OOB area size: 16 bytes
[    0.783365] sector size: 16 KiB
[    0.783845] pages number: 262144
[    0.784355] pages per sector: 32
[... 84 lines cut ...]
[    1.690755] r7 : b6f8c000  r6 : be8afd58  r5 : 011f8000  r4 : 0014f60c
[    1.691668] r3 : 00000005  r2 : 00000000  r1 : be8afb20  r0 : 00000000
[    1.692243] Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[    1.692677] Control: 10c5387d  Table: 4e9c4059  DAC: 00000015
[... cut ...]
[    1.717669] [<c0025534>] (do_exit) from [<c0026580>] (do_group_exit+0x48/0xa8)
[    1.718161] [<c0026580>] (do_group_exit) from [<c002e6dc>] (get_signal+0x42c/0x47c)
[    1.718644] [<c002e6dc>] (get_signal) from [<c00194a8>] (do_signal+0x8c/0x35c)
[    1.719604] [<c00194a8>] (do_signal) from [<c00198d8>] (do_work_pending+0x54/0xac)
[    1.720153] [<c00198d8>] (do_work_pending) from [<c0016c8c>] (work_pending+0xc/0x20)
[    1.721588] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[    1.721588]

y198nt, 2023-03-17:
and I don't know it end with [ 1.721588] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004. Normally when I use fat for .bin file, it always working successfully without any error, but when using fat for netgear router, there are a lot of error

extremecoders-re, 2023-03-17:
Yes it crashes with a kernel panic and that's why no network interfaces are detected. There is no generalized way to solve such issues. This will need more research to identify the reason for the panic.

The reason for the crash mioght be due to an invalid instruction.
```
FATAL: kernel too old
[ 1.675091] init (1): undefined instruction: pc=00010354
```

y198nt, 2023-03-24:
FATAL: kernel too old
[    0.832054] init (1): undefined instruction: pc=00010354
[    0.835649] CPU: 0 PID: 1 Comm: init Tainted: G        W       4.1.17+ #10
[    0.838057] Hardware name: Generic DT based system
[    0.838791] task: cf813ac0 ti: cf822000 task.ti: cf822000
[    0.841657] PC is at 0x10354
[    0.844627] LR is at 0x76afc
[    0.846653] pc : [<00010354>]    lr : [<00076afc>]    psr: 60030010
[    0.846653] sp : bea03c30  ip : 00000000  fp : bea03d90
[    0.853298] r10: 001186a0  r9 : 00000002  r8 : 00000001
[    0.854732] r7 : b6fca000  r6 : bea03d58  r5 : 00aff000  r4 : 0014f60c
[    0.854883] r3 : 00000005  r2 : 00000000  r1 : bea03b20  r0 : 00000000
[    0.856218] Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[    0.857720] Control: 10c5387d  Table: 4ea64059  DAC: 00000015
[    0.858753] Code: e3530004 1a000002 e3a03005 e584300c (e7f000f0) 
[    0.861320] potentially unexpected fatal signal 4.
[    0.864227] CPU: 0 PID: 1 Comm: init Tainted: G        W       4.1.17+ #10
[    0.864355] Hardware name: Generic DT based system
[    0.864445] task: cf813ac0 ti: cf822000 task.ti: cf822000
[    0.864710] PC is at 0x10354
[    0.866212] LR is at 0x76afc
[    0.867395] pc : [<00010354>]    lr : [<00076afc>]    psr: 60030010
[    0.867395] sp : bea03c30  ip : 00000000  fp : bea03d90
[    0.867961] r10: 001186a0  r9 : 00000002  r8 : 00000001
[    0.868117] r7 : b6fca000  r6 : bea03d58  r5 : 00aff000  r4 : 0014f60c
[    0.871191] r3 : 00000005  r2 : 00000000  r1 : bea03b20  r0 : 00000000
[    0.873053] Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[    0.873830] Control: 10c5387d  Table: 4ea64059  DAC: 00000015
[    0.874562] CPU: 0 PID: 1 Comm: init Tainted: G        W       4.1.17+ #10
[    0.875103] Hardware name: Generic DT based system
[    0.877357] [<c001c8dc>] (unwind_backtrace) from [<c0019c70>] (show_stack+0x10/0x14)
[    0.883679] [<c0019c70>] (show_stack) from [<c002e6cc>] (get_signal+0x41c/0x47c)
[    0.884918] [<c002e6cc>] (get_signal) from [<c00194a8>] (do_signal+0x8c/0x35c)
[    0.885866] [<c00194a8>] (do_signal) from [<c00198d8>] (do_work_pending+0x54/0xac)
[    0.886743] [<c00198d8>] (do_work_pending) from [<c0016c8c>] (work_pending+0xc/0x20)
[    0.900455] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[    0.900455] 
[    0.904947] CPU: 0 PID: 1 Comm: init Tainted: G        W       4.1.17+ #10
[    0.908241] Hardware name: Generic DT based system
[    0.914382] [<c001c8dc>] (unwind_backtrace) from [<c0019c70>] (show_stack+0x10/0x14)
[    0.914833] [<c0019c70>] (show_stack) from [<c0410938>] (panic+0x70/0x1c8)
[    0.915003] [<c0410938>] (panic) from [<c0025534>] (do_exit+0x3c0/0x774)
[    0.916071] [<c0025534>] (do_exit) from [<c0026580>] (do_group_exit+0x48/0xa8)
[    0.916268] [<c0026580>] (do_group_exit) from [<c002e6dc>] (get_signal+0x42c/0x47c)
[    0.916410] [<c002e6dc>] (get_signal) from [<c00194a8>] (do_signal+0x8c/0x35c)
[    0.918039] [<c00194a8>] (do_signal) from [<c00198d8>] (do_work_pending+0x54/0xac)
[    0.918967] [<c00198d8>] (do_work_pending) from [<c0016c8c>] (work_pending+0xc/0x20)
[    0.919732] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[    0.919732] 
QEMU: Terminated
-----------------------------------------------------------------------
I still don't know why it keeps saying KERNEL TOO OLD although I've install kernel 6.0.0-060000-generic version

extremecoders-re, 2023-03-24:
> I still don't know why it keeps saying KERNEL TOO OLD although I've install kernel 6.0.0-060000-generic version

The error message is about the [firmadyne kernel](https://github.com/firmadyne/kernel-v4.1). Not the kernel on the host OS. In addition I would also suggest looking at the address `00010354`. The instruction at that address is tripping qemu.

y198nt, 2023-03-24:
where is the kernel file which contain `00010354` address

extremecoders-re, 2023-03-24:
> where is the kernel file which contain `00010354` address

zImage.armel in https://github.com/firmadyne/kernel-v4.1/releases/tag/v1.1

sdahiasdba, 2024-03-31:
┌──(ayman㉿kali)-[~/Downloads/firmware-analysis-toolkit/firmadyne]
└─$ sudo ./fat.py Gaw5.6T02-4-DL-R1B020-ME.EN_upgrade\(0322113411\).en_upgrade                                      

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: Gaw5.6T02-4-DL-R1B020-ME.EN_upgrade(0322113411).en_upgrade
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/home/ayman/Downloads/firmware-analysis-toolkit/firmadyne/./fat.py", line 172, in <module>
    main()
  File "/home/ayman/Downloads/firmware-analysis-toolkit/firmadyne/./fat.py", line 160, in main
    image_id = run_extractor(args.firm_path)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ayman/Downloads/firmware-analysis-toolkit/firmadyne/./fat.py", line 54, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 432, in expect_exact
    return exp.expect_loop(timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
           ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fd3f235c050>
command: /usr/bin/sudo
args: ['/usr/bin/sudo', '--', '/home/ayman/Downloads/firmware-analysis-toolkit/sources/extractor/extractor.py', '-np', '-nk', 'Gaw5.6T02-4-DL-R1B020-ME.EN_upgrade(0322113411).en_upgrade', '/home/ayman/Downloads/firmware-analysis-toolkit/images']
buffer (last 100 chars): b''
before (last 100 chars): b'ayman/Downloads/firmware-analysis-toolkit/sources/extractor/extractor.py: command not found\r\nayman\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 19646
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Tag: '

help please

extremecoders-re, 2024-03-31:
@sdahiasdba Please create a new issue with a link to the firmware. In general, not all firmware will work right out of the box. Here it fails right on the extraction step, so the issue should be in the extraction part.

sdahiasdba, 2024-03-31:
The firmware that was used is http://www.downloads.netgear.com/files/GDC/WNAP320/WNAP320%20Firmware%20Version%202.0.3.zip and it has been extracted previously using the binwalk tool.

adi0x90, 2026-09-25:
The log shows the RAX30 image being skipped as `image/g3fax`, then `fat.py` waiting for a `Tag:` that never arrives.

FAT 2 has replaced that wrapper with native extraction and Binwalk/unblob fallbacks. There are successful Netgear extractions in the test log, but none confirm `RAX30-V1.0.9.92_1.img`. I don't want to mark RAX30 working based on another model.

Please try that exact image with the [v2 extraction workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime) and share its hash, `fat --version`, command, and output. That'll show whether there's still a format issue to fix.

adi0x90, 2026-09-26:
I retested the exact `RAX30-V1.0.9.92_1.img` from your log. Its MD5 matches `49241d7f2502d60e14e8f6005116fe3d`, and the published 2.0.1 Apple Silicon binary now recovers its rootfs with 2,134 regular files and symlinks on case-sensitive APFS. The earlier content comparison also matched all 2,134 entries against independent reference extraction.

Closing this original extraction report as fixed by [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108) and available in [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1). I'm tracking the kernel/init panic discussed later in this thread in [#110](https://github.com/attify/firmware-analysis-toolkit/issues/110), where guest boot still needs a fresh check.

This image has kernel-module filenames that differ only by case, so the project needs case-sensitive storage, such as Linux ext4 or case-sensitive APFS on macOS. The default Mac filesystem produces a partial extraction at that filename collision.

Please retry in a fresh FAT project, keeping the executable and its matching `share` directory together. For extraction, run `fat extract <fresh-project> --extractor native --json`. If it still fails, reply with `fat --version`, the firmware hash, and the JSON output so I can compare it with the tested image.

closed 2026-09-26 by adi0x90