attify/firmware-analysis-toolkit #89: EXT2-fs error (device sda1)

3826546, 2023-04-10:
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
[    7.420000] mount_root: mounting /dev/root
[    7.432000] mount_root: loading kmods from internal overlay
[    7.452000] mount_root: failed to launch kmodloader from internal overlay
[    7.552000] block: attempting to load /etc/config/fstab
[    7.560000] block: unable to load configuration (fstab: Entry not found)
[    7.568000] block: no usable configuration
[    7.688000] firmadyne: sys_reboot[PID: 1 (procd)]: magic1:fee1dead, magic2:28121969, cmd:0
[    7.700000] procd: - early -
block: unable to load configuration (fstab: Entry not found)
block: no usable configuration
[    8.532000] procd: - ubus -
[    8.624000] procd: - init -
Please press Enter to activate this console.
[   10.408000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8426
[   10.420000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8427
[   10.428000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8428
[   10.444000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8430
[   10.452000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8431
[   10.496000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   10.516000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   10.536000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   10.616000] EXT2-fs error (device sda1): ext2_free_blocks: bit already cleared for block 38912
[   11.072000] EXT2-fs error (device sda1): ext2_free_blocks: bit already cleared for block 55297
[   11.176000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   11.320000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   13.684000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   15.368000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   15.896000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   22.496000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   29.548000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   35.684000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
[   35.904000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206

extremecoders-re, 2023-04-11:
Unfortunately not all firmware will work out of the box. Manual tweaking may be necessary.

jollytrivedi2019, 2023-06-21:
Same error

mkdir: cannot create directory '/var/run': Input/output error
[    2.884000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 1517
mkdir: cannot create directory '/var/log': Input/output error
[    2.892000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 1518
mkdir: cannot create directory '/var/lock': Input/output error
[    2.916000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 1519
mkdir: cannot create directory '/var/state': Input/output error
[    2.928000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 1520
mkdir: cannot create directory '/tmp/.uci': Input/output error
[    2.952000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 1520

adi0x90, 2026-09-25:
The deleted-inode errors and I/O failures while creating runtime directories can stop the guest from starting properly. I still need a matching run showing the filesystem is usable before calling this fixed.

FAT 2 has kernel profiles and artifact commands under `fat kernel`, along with a different native image-preparation path. A maintained kernel alone won't fix a damaged or incorrectly prepared filesystem.

Please share the firmware version/hash, backend/kernel profile, image-preparation output, and serial log from v2. I need those to tell whether this is still an extraction, filesystem-building, or boot problem.

adi0x90, 2026-09-26:
[PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108) fixes several extraction failures, but the ext2 errors here occur during emulation and have not been verified as fixed by that change.

Please share the firmware model/version, a download link or hash, and the backend/kernel used. With [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1), retry in a fresh project and attach the new serial log if the filesystem errors remain.

I'm keeping this open so the runtime filesystem problem can be checked against a known image.