attify/firmware-analysis-toolkit #91: error while boot with a new kernel file

y198nt, 2023-06-15:
I' m currently trying to emulate RAXE300 firmware (latest version),the kernel of RAXE300 require is > 4.19 version with 32-bit, arm architecture, btw i'm using the config.armel file from https://github.com/firmadyne/kernel-v4.1, after that I put the zImage in firmware-analysis-toolkit/firmadyne/binaries. But after running fat.py, there is nothing happen, literally, there is nothing in qemu.final.serial.log. I don't know if it came from the config.armel file or the tool does not recognize my kernel file. Can anyone help me with it. Thanks a lot <3

adi0x90, 2026-09-25:
A newer kernel can still produce no visible output if the QEMU machine, CPU, console, storage drivers, or boot arguments don't match. I'd want to check those alongside the kernel version.

FAT 2 has explicit kernel profiles and artifacts, `fat kernel` commands, and lookup settings including `FAT_SYSTEM_KERNEL_DIR`. That makes the kernel setup easier to inspect than replacing a `zImage` in the old Firmadyne directory, but I don't have a confirmed RAXE300 boot result here.

Please share the firmware version/hash, backend/kernel profile, launch configuration, and any serial output from v2 so I can tell whether boot is failing early or the output is going to another console.

adi0x90, 2026-09-26:
I tested public RAXE300 firmware `1.0.9.82` after [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108). Native extraction recovered a rootfs with 2,166 regular files and symlink targets matching independent reference extraction. These extraction checks used case-sensitive APFS.

Your report says “latest version,” so please share the exact release number or hash. The custom-kernel boot failure remains open: I haven't verified your kernel configuration or a successful RAXE300 boot.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

If the serial log is still empty, please include the kernel version/hash, QEMU version, backend, and launch output as well as the firmware identity.