attify/firmware-analysis-toolkit #92: ./setup.sh failed to execute

Jonty16117, 2023-06-18:
error log:

running install
running build
running build_scripts
creating build
creating build/scripts-3.10
copying and adjusting src/yaffshiv -> build/scripts-3.10
changing mode of build/scripts-3.10/yaffshiv from 644 to 755
running install_scripts
copying build/scripts-3.10/yaffshiv -> /usr/local/bin
changing mode of /usr/local/bin/yaffshiv to 755
running install_egg_info
Writing /usr/local/lib/python3.10/dist-packages/yaffshiv-0.1.egg-info
+ rm -rf yaffshiv
+ install_jefferson
+ git clone --quiet --depth 1 --branch master https://github.com/sviehb/jefferson
+ cd jefferson
+ /usr/bin/python3 -mpip install -r requirements.txt
Collecting cstruct==2.1
  Downloading cstruct-2.1-py2.py3-none-any.whl (15 kB)
Collecting python-lzo==1.14
  Downloading python-lzo-1.14.tar.gz (14 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: python-lzo
  Building wheel for python-lzo (setup.py) ... done
  Created wheel for python-lzo: filename=python_lzo-1.14-cp310-cp310-linux_x86_64.whl size=23853 sha256=b430ae5b67b9d24136be8365168ff876794a4238080f3920b30b861ca39fba6c
  Stored in directory: /root/.cache/pip/wheels/24/b1/fd/54cba394c2d62468dc06b7508ee372209bb1541f3222719136
Successfully built python-lzo
Installing collected packages: python-lzo, cstruct
Successfully installed cstruct-2.1 python-lzo-1.14
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
+ /usr/bin/python3 setup.py install
/tmp/jefferson/setup.py:4: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import setup
running install
running build
running build_py
creating build
creating build/lib
creating build/lib/jefferson
copying src/jefferson/jffs2_lzma.py -> build/lib/jefferson
copying src/jefferson/rtime.py -> build/lib/jefferson
copying src/jefferson/__init__.py -> build/lib/jefferson
running build_scripts
creating build/scripts-3.10
copying and adjusting src/scripts/jefferson -> build/scripts-3.10
changing mode of build/scripts-3.10/jefferson from 644 to 755
running install_lib
creating /usr/local/lib/python3.10/dist-packages/jefferson
copying build/lib/jefferson/jffs2_lzma.py -> /usr/local/lib/python3.10/dist-packages/jefferson
copying build/lib/jefferson/rtime.py -> /usr/local/lib/python3.10/dist-packages/jefferson
copying build/lib/jefferson/__init__.py -> /usr/local/lib/python3.10/dist-packages/jefferson
byte-compiling /usr/local/lib/python3.10/dist-packages/jefferson/jffs2_lzma.py to jffs2_lzma.cpython-310.pyc
byte-compiling /usr/local/lib/python3.10/dist-packages/jefferson/rtime.py to rtime.cpython-310.pyc
byte-compiling /usr/local/lib/python3.10/dist-packages/jefferson/__init__.py to __init__.cpython-310.pyc
running install_scripts
copying build/scripts-3.10/jefferson -> /usr/local/bin
changing mode of /usr/local/bin/jefferson to 755
running install_egg_info
Writing /usr/local/lib/python3.10/dist-packages/jefferson-0.4.1.egg-info
+ rm -rf jefferson
+ install_ubireader
+ git clone --quiet --depth 1 --branch master https://github.com/jrspruitt/ubi_reader
warning: Could not find remote branch master to clone.
fatal: Remote branch master not found in upstream origin

WD-2711, 2023-07-20:
I also had this problem and it was because the master branch name of github was changed to "main". you should change like these, file in the `./firmware-analysis-toolkit/binwalk/deps.sh`.
[image]
Secondly, you should change `./firmware-analysis-toolkit/setup.sh` like this.
[image]
In the end, run `./firmware-analysis-toolkit/setup.sh`

Kong725, 2024-06-18:
> I also had this problem and it was because the master branch name of github was changed to "main". you should change like these, file in the `./firmware-analysis-toolkit/binwalk/deps.sh`. [...]

谢谢 已解决 非常有效

Royalton, 2024-12-19:
binwalk had a new release and now is a docker container so the newest patch here no longer fixes it either.

Jonty16117, 2024-12-22:
Should I close this issue then?

adi0x90, 2026-09-25:
Looking at the log, yaffshiv and Jefferson finish installing. The failure comes next, when the script tries to clone a `master` branch that no longer exists in `ubi_reader`. Thanks to everyone who worked through the fixes in this thread.

FAT 2 no longer runs the old Binwalk `deps.sh` or those yaffshiv, Jefferson, and `ubi_reader` checkout steps.

Closing this since that installer has been replaced. Please use the [v2 installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md); if the new installer fails, open an issue with its output so I can see where it stops.

closed 2026-09-25 by adi0x90