attify/firmware-analysis-toolkit #93: /etc/rc.common: eval: 1: cannot create /proc/sys/kernel/hostname: Directory nonexistent

jollytrivedi2019, 2023-06-21:
I am getting this error on extracting using FAT on VM with Ubuntu 22.04

adi0x90, 2026-09-25:
This message shows a startup script trying to write the hostname through `/proc/sys/kernel/hostname`, but the path isn't there. I need more of the startup output to tell whether procfs was mounted or something failed earlier.

FAT 2 uses a different runtime setup, though a change to host networking wouldn't fix a missing guest `/proc` path on its own.

Please add the firmware model/version, exact command, and output leading up to this error. If this happens during v2 emulation, include the backend and serial log too.

adi0x90, 2026-09-26:
The missing `/proc/sys/kernel/hostname` path is a guest-startup problem, and this report does not identify the firmware image. [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108) fixes extraction; I haven't tested this guest-startup failure yet.

Please share the model/version and a firmware download link or hash. If you retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1), use a fresh project and include the backend, kernel, and full serial log. I'll keep this open pending that image-specific check.