attify/firmware-analysis-toolkit #94: Cant ping from the Emulated to client

AnduinBrian, 2023-12-21:
Hello i use FAT to emulate the WR841N device.
After i login, i got shell root. I start iptables to accept the connect:
```
iptables -F
iptables -P INPUT ACCEPT
iptables -P OUTPUT ACCEPT
iptables -P FORWARD ACCEPT
```
I can connect to web server from client and the client can ping the emulated router.
But on the router, i cant ping to client (192.168.0.2)
[image]
But on the client, i can ping to the emulated router.
[image]

I can ping from router to client when i emulated the NetGear D6000. So what actually happend in my case ?

extremecoders-re, 2023-12-22:
Is there a firewall on the host which is preventing ping? 
In case if you already are able to achieve bidirectional communication it shouldn't be an issue.

Also please run `ifconfig`, `brctl` within the router and check the interfaces. Maybe there is no route or a misconfigured route which is preventing traffic from reaching the host.

AnduinBrian, 2023-12-25:
My ufw is off.
There is no bidirectional communication because i cant connect to my tftp sv on the 0.2 client.
The router ifconfig
[image]
The client ifconfig
[image]
I tried to set the broadcast same as the router but still cant.

AnduinBrian, 2023-12-25:
Found out, the IP range 192.168.0.1/24 is the problem.
I manualy delete the bridge and create a new one in the router (using `brctl`). After that i use `ifconfig` to set ip on both.
It worked.

closed 2023-12-25 by AnduinBrian