attify/firmware-analysis-toolkit #95: No IP show while emulating Tp-Link Archer AX50

y198nt, 2024-04-08:
You can download the file at here: https://us.softpedia-secure-download.com/dl/68da52af83d3218977ededbd17bcf150/6613a902/301008453/drivers/router/Archer%20AX50(EU)_V1_210730.zip
-------------------------------------------
When I run  sudo python3 fat.py archer-ax53/ax50v1_intel-up-ver1-0-11-P1\[20210730-rel54485\]_signed.bin. First there is no IP for interaction 
[image]

But after I pressed enter, it working normally. But it just there is no ip to interact 
Starting firmware emulation... use Ctrl-a + x to exit
.
[    0.000000] Linux version 2.6.39.4+ (ddcc@ddcc-virtual) (gcc version 5.3.0 (GCC) ) #2 Tue Sep 1 18:08:53 EDT 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 00678000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f897000 @ 00768000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 60672 bytes for tracking 1896 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 80702800, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252264k/254556k available (4554k kernel code, 2292k reserved, 1609k data, 240k init, 0k highmem)
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 333.33 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.004000] Calibrating delay loop... 2662.40 BogoMIPS (lpj=5324800)
[    0.032000] pid_max: default: 32768 minimum: 301
[    0.036000] Mount-cache hash table entries: 512
[    0.044000] Performance counters: No available PMU.
[    0.052000] NET: Registered protocol family 16
[    0.064000] bio: create slab <bio-0> at 0
[    0.064000] vgaarb: loaded
[    0.068000] SCSI subsystem initialized
[    0.068000] libata version 3.00 loaded.
[    0.072000] usbcore: registered new interface driver usbfs
[    0.072000] usbcore: registered new interface driver hub
[    0.072000] usbcore: registered new device driver usb
[    0.076000] pci 0000:00:00.0: [11ab:4620] type 0 class 0x000600
[    0.076000] pci 0000:00:00.0: reg 14: [mem 0x01000000-0x01ffffff pref]
[    0.080000] pci 0000:00:0a.0: [8086:7110] type 0 class 0x000601
[    0.080000] pci 0000:00:0a.1: [8086:7111] type 0 class 0x000101
[    0.080000] pci 0000:00:0a.1: reg 20: [io  0x0000-0x000f]
[    0.084000] pci 0000:00:0a.2: [8086:7112] type 0 class 0x000c03
[    0.084000] pci 0000:00:0a.2: reg 20: [io  0x0000-0x001f]
[    0.084000] pci 0000:00:0a.3: [8086:7113] type 0 class 0x000680
[    0.088000] pci 0000:00:0a.3: address space collision: [io  0x1100-0x110f] conflicts with GT-64120 PCI I/O [io  0x1000-0x1fffff]
[    0.088000] pci 0000:00:12.0: [1013:00b8] type 0 class 0x000300
[... 215 lines cut ...]
[    1.964000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: cat: can't open '/proc/cmdline': No such file or directory
[    2.088000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.088000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
[    2.112000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.112000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
[    2.132000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.132000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
[    2.152000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.152000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
[    2.172000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.172000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
[    2.192000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.192000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
[    2.208000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
[    2.212000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 464
sysinit: sysctl: chdir(/proc/sys): No such file or directory
QEMU: Terminated
---------------------------------------------------

y198nt, 2024-04-08:
[qemu.final.serial.log](https://github.com/attify/firmware-analysis-toolkit/files/14904144/qemu.final.serial.log)

it still error although i raise the time up to 180, 
sudo python3 fat.py archer-ax53/_ax50v1_intel-up-ver1-0-11-P1\[20210730-rel54485\]_signed.bin.extracted/0x460AA/460AA.squashfs

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: 460AA.squashfs
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /home/gh0st/firmware-analysis-toolkit/firmadyne/scratch/
2/run.sh
Starting firmware emulation... use Ctrl-a + x to exit

justin-in-time, 2024-08-27:
I'm having the same issue with the firmware I'm looking at.  No network interface is coming up and getting commands not found . I increased my timeout as well.  Is it not moving over the squashfs-root directory because it seems like the filesystem isn't working.

[    8.844000] OFFSETS: offset of pid: 0x100 offset of comm: 0x1f0
grep: /proc/filesystems: No such file or directory
sh: 1: unknown operand
/etc/rc.common: line 85: can't open /proc/cmdline: no such file
netlink_bind socket failed

Try `iptables -h' or 'iptables --help' for more information.
iptables: Index of insertion too big.
iptables v1.4.10: option `-o' requires an argument
Try `iptables -h' or 'iptables --help' for more information.
/sbin/uci: Entry not found (cloud_status.client_info.connect_status=0)
/sbin/uci: Entry not found (cloud_status.client_info.fw_download_progress=0)
/sbin/uci: Entry not found (cloud_status.client_info.fw_download_status=0)
/sbin/uci: Entry not found (cloud_status.client_info.fw_verify_status=0)
/sbin/uci: Entry not found (cloud_status.client_info.disconnect_reason=0)
/sbin/uci: Entry not found (cloud_status.client_info.reconnect_time=0)
uci: Entry not found (cloud_status)

Previous examples show this working on netgear firmware. Is there additional effort that needs to be made to develop this capability for other vendors? It would be nice to get this to work with easy.

adi0x90, 2026-09-25:
The AX50 boot log has ext2 deleted-inode errors and missing `/proc/sys` paths. I'd start there: those problems can stop the guest before it reaches working networking or a web service.

FAT 2's native `qemu-direct` system mode uses port forwarding, but that doesn't fix a guest filesystem or startup failure.

Please share the image hash and v2 extraction, preflight, and serial logs for the same firmware. I still need to see it prepare and boot successfully before calling the missing-interface problem fixed.

adi0x90, 2026-09-26:
The named Archer AX50 firmware, `ax50v1_intel-up-ver1-0-11-P1[20210730-rel54485]_signed.bin`, now extracts with the native engine after [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108). All 5,155 regular files and symlink targets matched independent reference extraction. These extraction checks used case-sensitive APFS.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

I'm keeping this open for the boot/network problem. Please attach a fresh serial log and include the backend and kernel used, especially if the `/proc` or ext2 errors still appear. Extraction is verified for this release; a working guest network and web interface are still to be checked.