attify/firmware-analysis-toolkit #99: sed: can't read deps.sh: No such file or directory

Anu9916, 2025-02-24:
**after running **./setup.sh**  shows error:**  

 cdac@host-cdac:~/Desktop/securityDomain$ git clone https://github.com/attify/firmware-analysis-toolkit

cd firmware-analysis-toolkit

./setup.sh

Cloning into 'firmware-analysis-toolkit'...

remote: Enumerating objects: 159, done.

remote: Counting objects: 100% (159/159), done.

remote: Compressing objects: 100% (86/86), done.

remote: Total 159 (delta 82), reused 141 (delta 71), pack-reused 0 (from 0)

Receiving objects: 100% (159/159), 47.33 KiB | 1.28 MiB/s, done.

Resolving deltas: 100% (82/82), done.

Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease

Hit:2 http://in.archive.ubuntu.com/ubuntu jammy InRelease

Hit:3 http://in.archive.ubuntu.com/ubuntu jammy-updates InRelease

Hit:4 http://in.archive.ubuntu.com/ubuntu jammy-backports InRelease

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

57 packages can be upgraded. Run 'apt list --upgradable' to see them.

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

fakeroot is already the newest version (1.28-1ubuntu1).

python3-pexpect is already the newest version (4.8.0-2ubuntu1).

uml-utilities is already the newest version (20070815.4-1).

vlan is already the newest version (2.0.5ubuntu5).

busybox-static is already the newest version (1:1.30.1-7ubuntu3.1).

kpartx is already the newest version (0.8.8-1ubuntu1.22.04.4).

qemu-system-arm is already the newest version (1:6.2+dfsg-2ubuntu6.24).

qemu-system-mips is already the newest version (1:6.2+dfsg-2ubuntu6.24).

qemu-system-x86 is already the newest version (1:6.2+dfsg-2ubuntu6.24).

qemu-utils is already the newest version (1:6.2+dfsg-2ubuntu6.24).

snmp is already the newest version (5.9.1+dfsg-1ubuntu2.7).

tar is already the newest version (1.34+dfsg-1ubuntu0.1.22.04.2).

unzip is already the newest version (6.0-26ubuntu3.2).

util-linux is already the newest version (2.37.2-4ubuntu3.4).

wget is already the newest version (1.21.2-2ubuntu1.1).

python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).

0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.

**Installing binwalk**

**Cloning into 'binwalk'...**

remote: Enumerating objects: 302, done.

remote: Counting objects: 100% (302/302), done.

remote: Compressing objects: 100% (258/258), done.

remote: Total 302 (delta 65), reused 129 (delta 37), pack-reused 0 (from 0)

Receiving objects: 100% (302/302), 5.44 MiB | 3.76 MiB/s, done.

Resolving deltas: 100% (65/65), done.

**sed: can't read deps.sh: No such file or directory**

extremecoders-re, 2025-02-24:
The issue happens because upstream binwalk has been rewritten from scratch in Rust.
A temporary work around would be to edit setup.sh and use a pinned version in the clone command, before the rewrote it in Rust.

https://github.com/attify/firmware-analysis-toolkit/blob/872d48e2598ef5eea6e46c8a502090d874b9c0ff/setup.sh#L13

Something like
```
git clone --depth 1 --branch v2.3.4 https://github.com/ReFirmLabs/binwalk
```

Anu9916, 2025-02-25:
ok thanks **extremecoders-re**

adi0x90, 2026-09-25:
The old installer expected `deps.sh` in the Binwalk checkout. That file disappeared when Binwalk was rewritten in Rust, which is why the `sed` step failed.

FAT 2's [installer](https://github.com/attify/firmware-analysis-toolkit/blob/master/scripts/install.sh) uses Cargo when it needs to install Binwalk, with `3.1.0` as the default version. It can also reuse an existing Binwalk that passes the version check. There's no `deps.sh` patching step anymore.

Closing this since the new installer handles the Rust version directly. The older Binwalk pin suggested in this thread is no longer needed for v2. Thanks for sharing the workarounds while this was broken.

closed 2026-09-25 by adi0x90